olivergregorius/sun2000_modbus #16: Step By Step

salvifran78, 2022-09-15:
Hi Oliver,
i have a sun2000 inverter. I try to use your program but i need support. You can heko me with a detailed installation step?
Thank you for your help.

olivergregorius, 2022-09-15:
Hi @salvifran78, how far did you come so far?

salvifran78, 2022-09-16:
Hi Oliver, 
because i'am not an expert of python, i used yaml configuration with modbus on home assistant. I found some interpeter that work for me. If you like, i can send you my yaml file.

#

olivergregorius, 2022-09-16:
This project is a library for being used in some Python applications, providing an easy to use interface to the internal Modbus interface of the Sun2000 inverter. It is not a standalone application that can be used out of the box.
I'm not familiar with Home Assistant, I've heard and read about it recently, but I don't have any experience with that software. But as far as I know, what Home Assistant is and can do, this library does not seem to be usable for your use case.

olivergregorius, 2022-10-06:
Closing this due to inactivity.

closed 2022-10-06 by olivergregorius