olivergregorius/sun2000_modbus #34: Upgrade to v3 of "Solar Inverter Modbus Interface"

Rosi2143, 2024-03-23:
Hi,

there is a [new release](https://community.symcon.de/uploads/short-url/pqZXWOienoBK2AsEzGD2oH1bcPR.pdf) V3 available.

Are there activities to update the library to the new version?

I will get a new installation of the SUN/LUNA 2000 soon and would be interrested in using/contributing to this great project.

olivergregorius, 2024-04-03:
Hi @Rosi2143 , I will check that soon.

olivergregorius, 2025-12-06:
Hi @Rosi2143 , some time has passed but finally I've prepared a new beta-version of this library containing all register changes up to version 9 of the Modbus register definitions. You will find more information in #49 

The beta version you could try is 3.0.0-b2.