olivergregorius/sun2000_modbus #38: Not working after today's update

Jesus-M, 2024-11-25:
After I upgraded to the version 2.3.0 today it stopped working. 

I have a very simple code, and I have even simplified it to make almost a copy of the example, and this is the result:

```
#!/usr/bin/env python3

from sun2000_modbus import inverter
from sun2000_modbus import registers

inverter = inverter.Sun2000(host='172.16.0.198',port='6607')
inverter.connect()
if inverter.isConnected():
#   AccumulatedEnergyYield      = inverter.read_raw_value(registers.InverterEquipmentRegister.AccumulatedEnergyYield)
  x = inverter.read_formatted(registers.InverterEquipmentRegister.InputPower)
  print(x)

#   DailyEnergyYield            = inverter.read_raw_value(registers.InverterEquipmentRegister.DailyEnergyYield)
#   ActivePower                 = inverter.read_raw_value(registers.InverterEquipmentRegister.ActivePower)
#   DeviceStatus                = inverter.read_formatted(registers.InverterEquipmentRegister.DeviceStatus)
#   print("%d,%5.3f,%4.2f,%s" % (AccumulatedEnergyYield*10,DailyEnergyYield/100,ActivePower/1000,DeviceStatus))
```

and the result:

```
INFO:sun2000_modbus.inverter:Successfully connected to inverter
ERROR:sun2000_modbus.inverter:Inverter unit did not respond
Traceback (most recent call last):
  File "/cinemateka/scripts/pvrrd-local.py", line 11, in <module>
    x = inverter.read_formatted(registers.InverterEquipmentRegister.InputPower)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/sun2000_modbus/inverter.py", line 73, in read_formatted
    value = self.read(register)
            ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/sun2000_modbus/inverter.py", line 65, in read
    raw_value = self.read_raw_value(register)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/sun2000_modbus/inverter.py", line 57, in read_raw_value
    raise register_value
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
```

The first error says `ERROR:sun2000_modbus.inverter:Inverter unit did not respond`, but it responds to ping in the local network, the first message is `INFO:sun2000_modbus.inverter:Successfully connected to inverter`, and it is online and connected to the Huawei management system.

Any idea?

Thanks

closed 2024-11-26 by olivergregorius

olivergregorius, 2024-11-26:
@Jesus-M, thanks for your bug report. Indeed the last update was a breaking change as it changed the default value for `unit`/`slave` to 1. I reverted it back to 0 as this seems to be working for most of the users (including me). The fix is included in the latest [Release 2.3.1](https://github.com/olivergregorius/sun2000_modbus/releases/tag/v2.3.1). If someone needs to call another `slave`, it can be set explicitly in the constructor.

I also fixed undesired propagation of log messages to the root logger.

FYI @sg26565 tl;dr Upgrading to 2.3.1 may break your setup, please set `slave=1` explicitly in the inverter's constructor.

Jesus-M, 2024-11-26:
thank you for the quick fix!

(tested and working!)