olivergregorius/sun2000_modbus #42: [question] pymodbus requirement

Jesus-M, 2025-03-11:
The requirements file indicated `pymodbus==3.7.4` as a strict requirement.

Now it is available **pymodbus 3.8.6** which I require for other stuff, but cannot install because of the strict requirement for sun2000_modbus and I wonder if it would work with a higher release, changing the requirement to a 'minimum version'

olivergregorius, 2025-03-11:
Hi @Jesus-M I will check that and take care of it.

olivergregorius, 2025-03-11:
@Jesus-M I've created a beta-version for you to test: https://pypi.org/project/sun2000_modbus/2.5.0b0/

I kindly appreciate your feedback.

Jesus-M, 2025-03-13:
Apologies for the delay, yesterday was a difficult day.

I tried `sun2000_modbus-2.5.0b0` with `pymodbus-3.7.4` and it worked fine, with a small test script I use (just retrieving and printing a few variables):
```
# test.py DailyEnergyYield DeviceStatus
INFO:sun2000_modbus.inverter:Successfully connected to inverter
inverter.read_raw_value(registers.InverterEquipmentRegister.DailyEnergyYield)
inverter.read_raw_value(registers.InverterEquipmentRegister.DeviceStatus)
0.04
512.0
#
```

But when I upgraded to `pymodbus-3.8.6` it failed:
```
# test.py DailyEnergyYield DeviceStatus
INFO:sun2000_modbus.inverter:Successfully connected to inverter
inverter.read_raw_value(registers.InverterEquipmentRegister.DailyEnergyYield)
Traceback (most recent call last):
  File "/cinemateka/scripts/test.py", line 22, in <module>
    value = inverter.read(getattr(registers.InverterEquipmentRegister, arg))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/sun2000_modbus/inverter.py", line 67, in read
    raw_value = self.read_raw_value(register)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/sun2000_modbus/inverter.py", line 56, in read_raw_value
    register_value = self.inverter.read_holding_registers(register.value.address, register.value.quantity, slave=self.slave)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
#
```

not sure yet if the problem is with my script or with how `sun2000_modbus` calls `pymodbus`

olivergregorius, 2025-03-13:
Thanks for your reply @Jesus-M , then I will have to dig a little deeper. I will get back to you, soon.

olivergregorius, 2025-03-13:
@Jesus-M please test again with https://pypi.org/project/sun2000_modbus/2.5.0b2/

Jesus-M, 2025-03-14:
`sun2000_modbus-2.5.0b2` with `pymodbus 3.7.4` seems to work fine

```
# pip install sun2000_modbus==2.5.0b2
(...)
Successfully installed sun2000_modbus-2.5.0b2
# 
# test.py DailyEnergyYield DeviceStatus
INFO:sun2000_modbus.inverter:Successfully connected to inverter
inverter.read_raw_value(registers.InverterEquipmentRegister.DailyEnergyYield)
inverter.read_raw_value(registers.InverterEquipmentRegister.DeviceStatus)
0.01
512.0
#
```

`sun2000_modbus-2.5.0b2` with `pymodbus 3.8.6 `seems to work fine too

```
# pip install pymodbus --upgrade
(...)
Successfully installed pymodbus-3.8.6
# 
# test.py DailyEnergyYield DeviceStatus
INFO:sun2000_modbus.inverter:Successfully connected to inverter
inverter.read_raw_value(registers.InverterEquipmentRegister.DailyEnergyYield)
inverter.read_raw_value(registers.InverterEquipmentRegister.DeviceStatus)
0.01
512.0
#
```

thanks!!

Jesus-M, 2025-03-14:
Never used the writing functions before, I'd need to try to change something non-disruptive for a first go XDD

Jesus-M, 2025-03-14:
I cannot get it working, but maybe I don't know how to use the write function.

Just reading:
```
# test.py TimeZone
INFO:sun2000_modbus.inverter:Successfully connected to inverter
inverter.read_raw_value(registers.InverterEquipmentRegister.TimeZone) ->  0.0
#
```

Writing attempt:
```
# test-w.py
INFO:sun2000_modbus.inverter:Successfully connected to inverter
Reading:
  registers.InverterEquipmentRegister.TimeZone ->  0.0
Writing:
Traceback (most recent call last):
  File "/cinemateka/scripts/test-w.py", line 15, in <module>
    inverter.write(registers.InverterEquipmentRegister.TimeZone, 2)
  File "/usr/lib/python3.12/site-packages/sun2000_modbus/inverter.py", line 121, in write
    response = self.inverter.write_registers(address=register.value.address, values=chunks, slave=self.slave)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/pymodbus/client/mixin.py", line 514, in write_registers
    return self.execute(no_response_expected, pdu_reg.WriteMultipleRegistersRequest(address=address, registers=values,dev_id=slave))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/pymodbus/client/base.py", line 203, in execute
    return self.transaction.sync_execute(no_response_expected, request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/pymodbus/transaction/transaction.py", line 109, in sync_execute
    self.pdu_send(request)
  File "/usr/lib/python3.12/site-packages/pymodbus/transaction/transaction.py", line 168, in pdu_send
    packet = self.framer.buildFrame(self.trace_pdu(True, pdu))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/pymodbus/framer/base.py", line 62, in buildFrame
    data = message.function_code.to_bytes(1,'big') + message.encode()
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/pymodbus/pdu/register_message.py", line 221, in encode
    packet += struct.pack(">H", value)
              ^^^^^^^^^^^^^^^^^^^^^^^^
struct.error: required argument is not an integer
#
```

The code I am trying to use is:
```
#!/usr/bin/env python3

import sys
from sun2000_modbus import inverter
from sun2000_modbus import registers

inverter = inverter.Sun2000(host='172.16.0.198',port='6607')
inverter.connect()
if inverter.isConnected():
  print('Reading:')
  value = inverter.read(registers.InverterEquipmentRegister.TimeZone)
  print('  registers.InverterEquipmentRegister.TimeZone -> ', value)

  print('Writing:')
  inverter.write(registers.InverterEquipmentRegister.TimeZone, 2)
  print('Reading:')
  value = inverter.read(registers.InverterEquipmentRegister.TimeZone)
  print('  registers.InverterEquipmentRegister.TimeZone -> ', value)
else:
  sys.exit("** Connection to inverter failed")
```

olivergregorius, 2025-03-17:
@Jesus-M sorry for the delay. Could you please check with https://pypi.org/project/sun2000_modbus/2.5.0b3/ ?

Jesus-M, 2025-03-18:
it seems that is doing something more, but still failing.

Same code than before (except that the value used to write in TimeZone is the integer 2 instead of the float 2.0). There is an error returned (`Exception response 144 / 0`) and as seen in the second reading, the value is unaltered.
```
# test-w.py
INFO:sun2000_modbus.inverter:Successfully connected to inverter
Reading:
registers.InverterEquipmentRegister.TimeZone ->  0.0
Writing:
Exception response 144 / 0
Reading:
registers.InverterEquipmentRegister.TimeZone ->  0.0
#
```

olivergregorius, 2025-03-18:
@Jesus-M OK, the timezone register is set to and returns a value in minutes. Thus, that seems to be a valid error from the inverter as you are trying to offset the timezone by 2 minutes which doesn't make any sense ^^

The valid value range is defined as [-720, 840]. The value should be a multiple of 60. Could you please try setting a value of 60 or 120 or whatever?

Jesus-M, 2025-03-18:
since when reading it returns 0.0 (I am in UTC zone), I assumed it was hours, in any case, values should be multiple of 30, not 60, since there are a few official timezones with an offset of 30 minutes. Anyway, I tried now with 60 getting the same error:

```
# cat test-w.py
#!/usr/bin/env python3

import sys
from sun2000_modbus import inverter
from sun2000_modbus import registers

inverter = inverter.Sun2000(host='172.16.0.198',port='6607')
inverter.connect()
if inverter.isConnected():
  print('Reading:')
  value = inverter.read(registers.InverterEquipmentRegister.TimeZone)
  print('  registers.InverterEquipmentRegister.TimeZone -> ', value)

  print('Writing:')
  inverter.write(registers.InverterEquipmentRegister.TimeZone, 60)
  print('Reading:')
  value = inverter.read(registers.InverterEquipmentRegister.TimeZone)
  print('  registers.InverterEquipmentRegister.TimeZone -> ', value)
else:
  sys.exit("** Connection to inverter failed")
#
```
```
# test-w.py
INFO:sun2000_modbus.inverter:Successfully connected to inverter
Reading:
registers.InverterEquipmentRegister.TimeZone ->  0.0
Writing:
Exception response 144 / 0
Reading:
registers.InverterEquipmentRegister.TimeZone ->  0.0
#
```

Jesus-M, 2025-03-18:
maybe that is my problem as well, that I am not allowed to change thing in my inverter. As I said before, I never used the writing functions before.

To me, the current status is perfectly functional and it allows me to use now `pymodbus 3.8.6`, which was my original need. So, thanks a lot!

leggler, 2025-03-22:
hi - i followed your discussion. just wanted to let you know that I think enabling newer versions of pymodbus is a great improvement.

So I used your library to create a bridge (https://github.com/leggler/PV-Aggregator/blob/main/SetupManualInstructions.md) - reading out several Huawei sun2000 units, and providing a single signal to my Energymanagement-system on a modbus-tcp server using pymodbus. Now I can upgrade to newer pymodbus versions ;)

olivergregorius, 2025-03-24:
Hi @leggler , I appreciate your feedback. Keep up the good work 👍

PR #43 (2025-04-27): #42: Softened dependency requirements
Closes #42
closed 2025-04-27 by olivergregorius